tscircuit/tscircuit #94: Setup not working

kainpets, 2024-04-25:
Running setup following README returns the attached error.

Might be I'm doing something wrong but looks to me like the link to the npm registry isn't right? 
@tsci:registry=https://registry-api.tscircuit.com/npm

[image]

[image]
  
[image]

seveibar, 2024-04-25:
Taking a look!

seveibar, 2024-04-25:
I think this might have something to do with your globally installed `tsx` version, can you do `npx tsx --version`?

seveibar, 2024-04-25:
Hmm actually I think it might be do to the default package.json created by `npm`, working on a reproduction...

seveibar, 2024-04-25:
Can you paste your `package.json` file?

kainpets, 2024-04-25:
[image]

seveibar, 2024-04-25:
I think you need to add "type": "module" to the package.json, I've updated the latest version of `tsci init` to add it automatically

kainpets, 2024-04-25:
Yep, that worked, thank you! Should've probably realised I had to run `tsci init`. 😅

closed 2024-04-25 by kainpets

seveibar, 2024-04-26:
thanks for trying it out!!!!!