tscircuit/tscircuit #120: Ports on the top or bottom of bug render incorrectly

KorvinSzanto, 2024-05-05:
Here's a minimal example board:

```tsx
export const Foo = () => <bug port_arrangement={{top_size: 2}} port_labels={{1: 'foo', 2: 'baz'}} />
```

| Side | Example |
|--------|--------|
| Top | [image] |
| Bottom | [image] |
| Left | [image] |
| Right | [image] | 

Note that for left and right the red dashes line up with the blue boxes whereas for top and bottom they do not.

KorvinSzanto, 2024-05-05:
I noticed everything is upside down when using `<Schematic>...</Schematic>` from `@tscircuit/schematic-viewer1` compared to the orientation in `tsci dev`, perhaps that's related

seveibar, 2024-05-05:
Yea recently I flipped the Y axis so that positive was up, this has had a lot of second-order consequences. Will get this fixed, thanks for reporting!

seveibar, 2024-05-06:
Hey @KorvinSzanto , I _think_ this should be fixed now with @tscircuit/schematic-viewer@1.2.4 and @tscircuit/builder@1.5.88. Note that the builder is a peer dependency of the schematic-viewer.

[image]

closed 2024-05-31 by seveibar