tscircuit/tscircuit #208: Modify the "soup" specification to add Oval support

seveibar, 2024-06-25:
We'll need to support "circle" (default) and "oval", also probably pill as shown in #205 

[image]

imrishabh18, 2024-06-26:
Hey @seveibar! 👋 I would like to work on this.

seveibar, 2024-06-26:
Thanks to @imrishabh18 we now have support for Ovals, but we're still missing support for Pills, I'm going to edit the issue title to reflect that we only need "Pill" support now.

RE: Pills vs ovals

[image]

[image]

seveibar, 2024-06-26:
Looking into this more, I think our definition of Oval _might_ cover creating pills. I'm going to close for now.

seveibar, 2024-06-26:
Ah no we do need 2 different shapes, but the parameters outer/inner width/height i _think_ would be the same, the drawing function would just change based on the shape being used.

imrishabh18, 2024-06-28:
Where is the drawing function located, can you please point me to that?

seveibar, 2024-06-28:
@imrishabh18 this video should help!

https://github.com/tscircuit/tscircuit/assets/1910070/64c4c3c8-03e8-4ee1-ae96-5371dac93d0d

closed 2024-07-20 by imrishabh18