tscircuit/tscircuit #215: Pills not converted properly in kicad file conversion

seveibar, 2024-06-29:
- [ ] Update https://github.com/tscircuit/kicad-mod-converter to support outputting "pills" for some kicad shapes
- [ ] Update https://github.com/tscircuit/pcb-viewer to support rendering pills
- [ ] Update https://github.com/tscircuit/kicad-viewer dependencies to fix conversion & pcb-viewer

[image]

imrishabh18, 2024-07-01:
Before
[image]

After
[image]

seveibar, 2024-07-01:
From the image above, that's still not quite correct, the kicad version is actually a pill not an oval

imrishabh18, 2024-07-01:
Yeah, I will add the pill shape now. Just wanted to do an oval first, will add the pill shape after this

imrishabh18, 2024-07-04:
[image]

seveibar, 2024-07-04:
@imrishabh18 that's amazing!!!!

closed 2024-07-06 by imrishabh18