tscircuit/tscircuit #252: Create `@tscircuit/soup-to-svg` that can convert pcb or schematic elements into a PNG image

seveibar, 2024-07-14:
This would be extremely useful for snapshot testing, we could then have tests log a visual snapshot so that some things wouldn't break.

Recently plated holes were completely broken because the usage of [`Number` was converting values like `1mm` to `NaN`](https://github.com/tscircuit/builder/pull/73/files), we didn't catch this in part because we didn't have readable visual snapshots

CC @imrishabh18 might be an interesting one- lots of steps though!

- [ ] Introduce `@tscircuit/soup-to-svg`
- [ ] Create function `soupToPng` that converts the svg to a png
- [ ] Add `schematicSnapshot(soup)` as a return from `getTestFixture`
- [ ] Add `pcbSnapshot(soup)` as a return from `getTestFixture`
- [ ] Demonstrate that a test will show a visual diff in github when there's a change

> The `schematicSnapshot` and `pcbSnapshot` functions would write a `png` file `<test name>.snapshot.png`. When this
> png is changed on CI, the test will fail. This will allow us to fail PRs and review for visual differences

seveibar, 2024-07-14:
For what it's worth: I think Claude if prompted properly could build each function for converting soup to svg. The only somewhat difficult part IMO is making sure that the circuit is scaled properly (to e.g. a 20mmx20mm grid). To do this, you would want to use the `transformation-matrix` library, which has easy-to-use functions for translating and scaling coordinates.

imrishabh18, 2024-07-15:
I can take a look at this. So, first step would be to create a `soup-to-svg` package where I need write the logic to parse svg to soup, right?

seveibar, 2024-07-15:
There is also already a repo here: https://github.com/tscircuit/soup-to-svg

You can use `bun init` if you'd like. I can add the build system later (or you can copy it from another project- we use `tsup`) I would also recommend installing `@tscircuit/plop` and dropping in the biome.json and format/typecheck bot.

closed 2024-09-14 by imrishabh18