tscircuit/tscircuit #258: When a component is on the bottom layer, flip the 3d model upside down

seveibar, 2024-07-15:
# Currently what happens

[image]

# What's supposed to happen

[image]

[image]

# How to fix

- [ ] Write a test with a component with a `cadModel` defined and put the component on the bottom layer
   - [ ] You can use this as the `objUrl` for the `cadModel` (it's the model above) `"https://modelcdn.tscircuit.com/easyeda_models/download?uuid=6331f645d89e4b919bdba0cb4f3544ce&pn=C124379"`
- [ ] Look for where the `cad_component` is created inside of `ComponentBuilder`, it should have `rotation.y` set to `180` when it's on the bottom layer
- [ ] Use the debug server and `logSoup` to make sure it flips properly

seveibar, 2024-07-18:
@Slaviiiii are you still working on this? If not can you unassign yourself

Slaviiiii, 2024-07-19:
I could not understand what the problem was with it

Slaviiiii, 2024-07-19:
I can reassign if I understand how the codebase works further in

closed 2024-07-25 by seveibar