tscircuit/tscircuit #264: Configurable trace widths for nets

seveibar, 2024-07-18:
For my motor controller board, my trace widths are super thin but some of these traces must carry a lot of power. We need to allow nets to have configurable trace widths via

```tsx
<net traceWidth="0.5mm" />
```

[image]

imrishabh18, 2024-07-30:
@seveibar For this, shouldn't we just have `traceWidth` prop? How to decide between the range of `minTraceWidth`  and `maxTraceWidth` which thickness to choose.

seveibar, 2024-07-30:
@imrishabh18 sometimes the traceWidth needs to automatically expand to be as large as possible whenever possible. For this we need to use a "trace expansion algorithm". Most power traces we want to be very wide, but will need to become narrow to fit into or through certain parts of the circuit.

We don't currently have a trace expansion algorithm although I'm very close to one, might have it finished this week.

We don't need the algorithm if we support a ui to edit trace width inside of hints, to do this we would first need to add traceWidth to the <tracehint /> component

The pcb-viewer probably doesn't support varying the width right now, also i doubt some of our checks support it. Nonetheless it's pretty important for routing power automatically.

imrishabh18, 2024-07-30:
I see, so I will just start with the `traceWidth` right now. I think after your work on the algo is done, probably we can then add the min and max

seveibar, 2024-07-30:
@imrishabh18 fwiw its ok to implement parts of this w/o the algorithm or ui etc. we just have to take baby steps haha

closed 2024-08-15 by imrishabh18