tscircuit/tscircuit #268: Highlight other connected pads/traces when you hover over a pad

seveibar, 2024-07-19:
This allows you to see pads and traces that should be connected

ShiboSoftwareDev, 2024-08-30:
[image]
I just started working on this but why is the trace not showing in the pcb-viewer tests?

ShiboSoftwareDev, 2024-08-30:
and is there a specific behavior you expect, or just normal selection but for all connected pads/traces? @seveibar

seveibar, 2024-08-30:
@ShiboSoftwareDev it might have been broken by recent updates to the trace renderer (CC @imrishabh18) it's an issue that we should fix.

@ShiboSoftwareDev hovering over a pad should change _the color of the pad_ to white, and all other connected pads/traces should also change to white. I don't think we should pop up the squares.

ShiboSoftwareDev, 2024-09-03:
[image]
[image]
@seveibar how is it looking so far?

seveibar, 2024-09-03:
@ShiboSoftwareDev thats great

closed 2024-09-12 by ShiboSoftwareDev