tscircuit/tscircuit #322: Spectra dsn import/export

NavinF, 2024-07-31:
Are there any plans to support dsn -> tscircuit soup -> dsn? AFAIK that's the de facto workflow for EDA tools and necessary for a future tscircuit autorouter to be useful

seveibar, 2024-07-31:
@NavinF yes we have started work here and we'll probably complete it very soon.

This repo has the converter: https://github.com/tscircuit/specctra-dsn-json

CC @andrii-balitskyi

seveibar, 2024-07-31:
@andrii-balitskyi this might be a good thread to discuss the progress on that project, are there any large remaining items? I know we got pretty far on parsing the dsn files within the freerouting project.

@imrishabh18 maybe you could also show an example of `circuit-to-png` w/ snapshot testing in that repo, that would probably help us track progress

andrii-balitskyi, 2024-08-01:
@seveibar There's a draft PR that adds support for converting DSN JSON into tscircuit soup. This PR was put on hold as our focus shifted to another project. Additionally, the parsing of DSN files hasn't been thoroughly tested, so there may be some inaccuracies in the produced DSN JSON files.

imrishabh18, 2024-08-01:
`specctra-dsn-json` lib doesn't give the soup it only converts it to json, right?

seveibar, 2024-08-01:
@imrishabh18 yea it needs that draft PR for DSN JSON to tscircuit soup before we can do snapshots, but the code to convert to soup will still stay inside the `specctra-dsn-json` repo.

closed 2026-02-21 by github-actions[bot]