tscircuit/tscircuit #327: Add support for pcb_keepouts

seveibar, 2024-08-07:
- [ ] Introduce `<keepout />`
- [ ] Introduce `pcb_keepout` soup component [tscircuit/soup](https://github.com/tscircuit/soup)
- [ ] Introduce `keepoutProps` in [tscircuit/props](https://github.com/tscircuit/props)
- [ ] Create obstacles from keepout to avoid routing through it

seveibar, 2024-08-07:
CC @andrii-balitskyi we discussed this today, also CC'ing @imrishabh18 if you're looking for a big issue not a bad one to pick up

imrishabh18, 2024-08-10:
@seveibar I am confused here a bit, the `KeepoutBuilder` can be defined using this `defineNewComponent` function like for the `viaBuilder` here. Or, have it's own class defined?

[image]

seveibar, 2024-08-10:
If you can get `defineNewComponent` to work that's the "new" pattern!

closed 2024-09-08 by imrishabh18