tscircuit/tscircuit #328: Arduino Nano

seveibar, 2024-08-07:
https://github.com/tscircuit/arduino-nano

- [ ] Bootstrap project
- [ ] Find all parts + JLC part numbers
- [ ] Add connections between components

swastikCommits, 2025-08-15:
/attempt #328

naaa760, 2025-10-02:
/attempt #328
@seveibar 
please check it out!

jdhruv1503, 2026-02-25:
/attempt #328
Submitting PR #2338 for bounty. Circuit ready for production testing.

kira-autonoma, 2026-03-15:
/attempt #328

Complete Arduino Nano V3.0 circuit implementation with ATmega328P-AU, CH340G USB-serial, AMS1117-5.0 voltage regulator, crystals, LEDs, reset circuit, DTR auto-reset, ICSP header, pin headers, and full decoupling network. JLCPCB part numbers included.

PR: https://github.com/tscircuit/tscircuit/pull/2653

Bilal-Lodhi, 2026-05-12:
/attempt #338
I'm an AI-augmented developer. I've noted the requirements for the MCU, crystals, and capacitors. I'm starting a precision implementation of the Arduino Nano V3.0 now.

closed 2026-09-27 by github-actions[bot]