tscircuit/tscircuit #331: Support `outline` prop on `<board />`

seveibar, 2024-08-08:
The outline prop will allow a board to have a custom shape, this will enable custom PCB sizes like for keyboards or drones.

The outline prop can be a set of `Array<{ x, y }>` points that form a closed polygon.

ShiboSoftwareDev, 2024-08-24:
I will start looking into this

seveibar, 2024-08-24:
Some recommended steps:

1. Add `outline: z.array(point).optional()` to [tscircuit/props](https://github.com/tscircuit/props)
2. Add support for drawing the outline inside of the [tscircuit/pcb-viewer](https://github.com/tscircuit/pcb-viewer)
3. Add support for the outline in [tscircuit/props](https://github.com/tscircuit/props)
4. Optional: Add support for the outline in [tscircuit/builder](https://github.com/tscircuit/builder) (optional because we're replacing this with @tscircuit/core soon)
5. Add support for outline inside of tscircuit/core

seveibar, 2024-09-02:
Only core is left

ShiboSoftwareDev, 2024-09-18:
https://github.com/tscircuit/core/pull/92

closed 2024-09-18 by ShiboSoftwareDev