tscircuit/tscircuit #363: Schematic Viewer missing pin lines

seveibar, 2024-08-22:
[image]

Abse2001, 2024-08-24:
I looked into the problem it seems that if you dont add the schPortArrangement prop on the bug/chip it will not draw the pin lines
[image]
[image]

seveibar, 2024-08-24:
oh weird, nice find

Abse2001, 2024-08-24:
Is it not meant to be like that?

seveibar, 2024-08-24:
Yea a chip should always have those red lines coming out connecting the ports, even if `schPortArrangement` isn't defined

seveibar, 2024-08-24:
I'm not sure if the bug is in the [tscircuit/builder](https://github.com/tscircuit/builder) or [tscircuit/schematic-viewer](https://github.com/tscircuit/schematic-viewer)

seveibar, 2024-08-24:
The way to determine what project the issue is in is probably to look at the circuit json/soup output and see if when `schPortArrangement` is defined if the soup changes and how. I think it's _probably_ a bug in [tscircuit/builder](https://github.com/tscircuit/builder)

Abse2001, 2024-08-24:
will look more into this problem

closed 2024-12-14 by Abse2001