tscircuit/tscircuit #443: Hackoberfest Preparation

seveibar, 2024-09-16:
Hackoberfest is coming! To prepare, we're going to need to create some epics and good "getting started" work for new tscircuit contributors.

There are three main projects we'd like people to work on:

- Our footprint library [tscircuit/footprinter](https://github.com/tscircuit/footprinter)
- Our 3d electronic components library [tscircuit/jscad-electronics](https://github.com/tscircuit/jscad-electronics)
- A yet-to-be-created sparkfun-clones library that clones sparkfun projects (they'll make great code examples for users!)

Thanks to @imrishabh18 for the idea!

RS-labhub, 2024-10-02:
waiting for it!

vickywane, 2024-10-03:
Looking forward to this!

@seveibar can we have a list of "good-first" issues somewhere?

imrishabh18, 2024-10-03:
Hey @vickywane! Thanks for the interest.

You can refer to this blog post, it has a lot of info about the issues, with the tutorials as well.

closed 2024-11-22 by seveibar