tscircuit/tscircuit #458: react doesn't like using key as a prop

ShiboSoftwareDev, 2024-10-04:
[image]

i found this issue in core and circuit-json-to-gerber repos but not in cli even tho it has a similar test

Yashwanth-Chandrakumar, 2024-10-18:
/assign

Yashwanth-Chandrakumar, 2024-10-18:
@ShiboSoftwareDev Hi Mate can you explain me how to reproduce this error and in which repo did you get it.

seveibar, 2024-10-18:
@ShiboSoftwareDev just seeing this but can confirm that this is annoying. I think it might happen when props is not defined.

@Yashwanth-Chandrakumar you could do a reproduction by creating a <diode /> component with a key prop inside of https://github.com/tscircuit/core

The docs directory explains how to create a test

ShiboSoftwareDev, 2024-10-18:
I think it happens because we have a prop called key when we shouldn't

Yashwanth-Chandrakumar, 2024-10-18:
@ShiboSoftwareDev Thanks Mate