tscircuit/tscircuit #461: Fix tscircuit/core to allow for nested subcircuits (`<group subcircuit pcbX={5} />` should shift everything in the group by 5)

imrishabh18, 2024-10-06:


imrishabh18, 2024-10-06:
@seveibar Can you explain more about this please?

seveibar, 2024-10-06:
Sure. There is a type of group called a `subcircuit`.

Within a `subcircuit`...
- All top-level names are unique (e.g. `name="C1", name="R1"`)
- Selectors do search across all components that aren't in a subcircuit. This is a tricky one to understand but here goes:

In this example the trace can connect C1 in R1, **this example WORKS**
```tsx
<group>
  <resistor name="R1" />
  <capacitor name="C1" />
  <trace from=".R1 .pin1" to=".C1 .pin1" />
</group>
```

In this example the trace CAN NOT connect C1 and R1, **this example DOES NOT WORK** and will throw an error that the trace couldn't find the component "C1"
```tsx
<group>
  <resistor name="R1" />
  <group subcircuit>
    <capacitor name="C1" />
  </group>
  <trace from=".R1 .pin1" to=".C1 .pin1" />
</group>
```

In this example the trace can connect C1 and R1, **this example WORKS AGAIN!**. The reason it works is the group has a name, so the selector knows it can go inside the subcircuit.

When you can't get inside a subcircuit without referencing the subcircuit, it's called the "subcircuit boundary"

```tsx
<group>
  <resistor name="R1" />
  <group subcircuit name="G1">
    <capacitor name="C1" />
  </group>
  <trace from=".R1 .pin1" to=".G1 .C1 .pin1" />
</group>
```

## Why are subcircuits important?

**Subcircuits allow you to have MODULES**. When we define higher level components like a `<Microprocessor />` we
use a subcircuit because a microprocessor contains a main chip along with a bunch of decoupling capacitors and supporting
chips to make it work (sometimes a power regulators or UART chip)

```tsx
export const Microcontroller = ({ name?: string }: { name: string }) => {
  return (
    <group subcircuit name={name}>
      <chip name="U1" />
      <resistor />
      <capacitor />
    </group>
  )
}
```

Subcircuits can be moved without impacting their routing

```tsx
<group subcircuit pcbX={5} pcbY={2}>
  {/* ... */}
</group>
```

imrishabh18, 2024-10-13:
@seveibar I was testing this out, but this seems to be already working.
https://github.com/user-attachments/assets/45475d79-0bf6-449d-ab04-041b786a84af

closed 2024-10-14 by imrishabh18