tscircuit/tscircuit #468: Allow configuring autorouter for subcircuits

seveibar, 2024-10-23:
We want to be very careful in how we enable this, tscircuit should generally assume that autorouting will be a "solved problem" in the future, so we don't want to have extremely autorouter-specific configuration that would mean upgrading tscircuit to better autorouters would result in regressions. For example, having `<group subcircuit autorouterConfig={{ maxIterations: 10 }} />` is too specific- future autorouters may not have a concept of an iteration.

We also want to be very cloud-compatible. I think it's inevitable that autorouting will be GPU-powered/dedicated chip powered in the future, and not everyone will have that in their local machines.

## Proposal: fast, quality, local, cloud

There are essentially 4 things that will change your experience. I think having these boolean props will be reasonably easy to deprecate

```tsx
// Default Fast
<group subcircuit autorouterQuality="fast" />

// Quality
<group subcircuit autorouterQuality="quality" />

// Default Inherit

// Local
<group subcircuit autorouterLocalOnly />
```

naaa760, 2024-10-24:
Hello @seveibar :)

I want to work on this issue.

seveibar, 2024-11-01:
@naaa760 we're changing this issue a bit with our freerouting integration via the introduction of an `autorouter` prop

This is a good issue for an early contributor: https://github.com/tscircuit/snippets/issues/151

closed 2026-02-20 by github-actions[bot]