tscircuit/tscircuit #476: Trace Constraints (decoupling capacitor traces should have max length)

seveibar, 2024-11-16:
- [ ] Add `maxLength` property to traces in [tscircuit/props](https://github.com/tscircuit/props)
- [ ] Add `maxDecouplingTraceLength` to `capacitor` in [tscircuit/props](https://github.com/tscircuit/props)
- [ ] Add constraint information to `source_trace` in circuit-json module
- [ ] Add `trace_length` as property of `pcb_trace` in circuit-json module
- [ ] Add a check inside  [tscircuit/checks](https://github.com/tscircuit/checks) for trace constraints
- [ ] Compute `pcb_trace.trace_length` in [tscircuit/core](https://github.com/tscircuit/core)
- [ ] Add `schematic_text` alongside a trace in [tscircuit/core](https://github.com/tscircuit/core) indicating it's length e.g. "1.2mm"

/bounty $100

This issue can be assigned to a frequent contributor (you should have at least one recent 2 star week on the [contribution tracker](https://github.com/tscircuit/contribution-tracker)

We may give some of the bounty, e.g. the first $50 dollars, after ~50% of the work has been done.

If you haven't already closed 3+ bounties, you're not eligible for this issue.

Abse2001, 2024-12-24:
/attempt #411

Abse2001, 2024-12-29:
/attempt #411

closed 2026-02-20 by github-actions[bot]