tscircuit/tscircuit #479: Autocloud Autorouting has some kind of inversion over either the X or Y axis

seveibar, 2024-11-20:
[image]

seveibar, 2024-11-22:
It doesn't look like this is caused by the registry api or freerouting cluster, we had some internal snapshot testing [here](https://github.com/tscircuit/registry-api/pull/35) that demonstrates that the api doesn't cause an inversion if you just send it circuit json

seveibar, 2024-11-22:
CC @imrishabh18 this is where i'm tracking the inversion issue. I'm a bit confused where it could be happening because the registry api test seems to imply it's NOT happening in any of the cloud stuff (dsn-converter, freerouting or registry-api). More testing needed...

seveibar, 2024-11-22:
It also seems to work for me for simple examples

Could it be an issue with something like plated holes? I'm not sure... the wifi on this plane is so bad i'm spending a lot of time waiting for things to load 😅

seveibar, 2024-11-22:
Just confirmed it only seems to invert plated holes. I have images but can't upload them over the plane wifi. Here's a simple example reproducing the inversion:

```tsx
export default () => (
  <board width="10mm" height="10mm" autorouter="auto-cloud">
    <resistor
      resistance="1k"
      footprint="axial"
      name="R1"
      schX={3}
      pcbX={3}
      pcbY={3}
      pcbRotation="90deg"
    />
    
    <capacitor
      capacitance="1000pF"
      footprint="0402"
      name="C1"
      schX={-3}
      pcbX={-3}
    />
    <trace from=".R1 > .pin1" to=".C1 > .pin1" />
  </board>
)
```

imrishabh18, 2024-11-22:
@seveibar The dsn file conversion from the circuit json for the plated_hole is the issue. The source trace is not getting converted to the correct `net` in dsn

imrishabh18, 2024-11-22:
@seveibar  Closing this as this fixed now

[image]

closed 2024-11-22 by imrishabh18