tscircuit/tscircuit #480: Autorouter with "auto-cloud" is only routing a single trace

seveibar, 2024-11-22:
[image]

CC @imrishabh18, my guess is this is in the registry-api, maybe we're not waiting for the job to finish? Needs investigation

Here's the reproduction:

https://tscircuit.com/editor?&snippet_type=board#data:application/gzip;base64,H4sIAEUCQWcAA62SvQ7CMAyEd57CygQDkAIj6YLEAzDBGNKURtA6Sl2BhHh3Qkn4lYCBLd/JlztL1geLjiDTuWx2BN0eiBS6HYDpGqXLYG8yKgRLeFkyKLTZFBRJNoQOG9JOsMu7r3bYZCz1Zm93uja1H2gJ4IqyUtrbtyyoOSJZZyr/J5/wUZQrWfqxRRK5VsVSHMengFat7zj8X96NP+W1uHqLV9JKZe75gUMBzrmd/1Ri9rJ0/6VF5JB7DScnlYbcYSnYYJFACgNrqoQBoRdmD0Ks++wYfXNMh+01pJ3eGWy5+44wAgAA

Abse2001, 2024-11-23:
@seveibar  it is routing two traces in schematic but only one in PCB what is you thought.
[image]

Abse2001, 2024-11-23:
and it is routing two traces if not the same port.
[image]
[image]

imrishabh18, 2024-11-23:
@seveibar Will take a look at this

imrishabh18, 2024-11-24:
@seveibar This is something related to the freerouting, as you can see in the freerouting for this circuit, we can see the white source trace connecting both of them to the `C1` pin, but autoroutes only one.   have tried editing this and changing the dsn file format, but freerouting is routing only one trace, when one pin is connected to two different pins.

https://github.com/user-attachments/assets/35b2818e-c135-4a25-ace4-ac0b525b41f6

**Claude's response:**

[image]

imrishabh18, 2024-11-25:
@seveibar Another thing, I have noticed is that if I am changing just the X position of `R2` on the circuit you have linked above. The freerouting is producing both the traces. 

https://tscircuit.com/editor?&snippet_type=board#data:application/gzip;base64,H4sIALA4RGcAA62RvQ7CIBSF9z7FDZMOKlVH6WLiAzjpiJRaou0l9DaaGN9dVPCvgw5ufCf38HGDPlp0BLkuZLsn6PVBZNBLAGYblC6Hg8mpFCzlVcWg1GZbUiTZEjpsSTvBrueB2mObs8yXfd3pxjR+4EYAd5S10r6+YyEtEMk6U/s7+ZSPY1zLyo8t08iNKlfiNDkHtGrzxNH/fA/u+tYdn5JWKvMUBg5Gzrld/GSdf2w5+FgzcvDe5eSk0lA4rAQbLlPIYGhNnTIg9MH8JYjPfW+MvzVmo9v3Z0k/uQBoZeeYIgIAAA==