tscircuit/tscircuit #482: Readme is outdated

imrishabh18, 2024-12-04:
The readme is outdate for this repo, the schematic screenshots or some of the other stuff like the table of core libraries is missing snippets.

There would be more things which needs to be updated according to your judgement

samyakshah3008, 2024-12-08:
Hey @imrishabh18 , I would like to work on this issue - here are my few enhancement ideas - 

We can reorder some sections (this is my opinion but would love to hear from everyone) - right after the links and our brand logo, we can have an introductory video of Severin. 

Also we should add Table of contents after our hero header, starter links and intro video. This will help to navigate very easily 💙

Sections: 

1. Hero header - which will have our brand logo with tagline React for Circuits
2. Starter links (we already have) but we can write them in formatted way and add a few more things like - commit activity pill eg. 65/month
3. Introductory video from Severin 
4. About Tscircuit - this will describe and give a brief content about tscircuit 
5. Example circuits section (this we have currently but can improve this how we have written and polish it more upon discussing with maintainers)
6. Community and next steps section where we can write about our community, our initiatives of giving bounties, talk a brief about our community meetings and write in short about what next steps we are planning
7. Getting Started section which we currently have (can move up too somewhere upon discussions)
8. Contact us section where we can write a brief about how can someone contact the team 
9. Developer setup guide (can write here or can move it to CONTRIBUTING.md file (currently we don't have but I think we should create it) , where we can write the steps a developer should follow to setup on their local 
10. Want to contribute to TSCircuit? You are most welcome section - We will directly add a link here for CONTRIBUTING.md file
11. Repo Activity (Can move up as well upon discussions) , I discovered - https://repobeats.axiom.co/ which generates an image with detailed analytics about the repo eg. https://repobeats.axiom.co/api/embed/50fd88e948fb4cc0a2d2041b8a8534a0dd7852e6.svg 
12. Development/Sub project org (we already have this but can polish it) 
13. FAQs (we have this but needs more polishing about the content)

seveibar, 2024-12-08:
Yea we should do enhancements incrementally (each one could be a PR for example), it really depends on the execution for almost all of these- but generally agreed these all sound great...

(8) is just discord I think, no need to have a specific section for that.

For point 11, this repo doesn't really have that much activity because it's an "aggregate package"- so I'm not 100% sure that would work on this particular repo (it could work on repos like core, we might even be able to list multiple repos in the beats? it's a cool image)

samyakshah3008, 2024-12-08:
> Yea we should do enhancements incrementally (each one could be a PR for example), it really depends on the execution for almost all of these- but generally agreed these all sound great...
>
> (8) is [...]

Cool! Noted the feedback for initial implementation plan, I'll be sending PR in chunks for the same to get feedbacks for each! 😄
Yes @seveibar , we can generate image for each repository ⚡if we want to combine in one image with multiple repos, from what I have checked it's not possible in one image but we can add multiple image in README of how we want. 💙

PR #484 (2024-12-11): [Docs]: Readme file Enhancements and add Contributing guide
## Description: 

- We currently have an outdated README file which needs enhancements and content polishings with few more things to add on.

## What changes are made?

## Section enhancement Checklist for README
- [x] Title
- [x] Starter Links
- [x] Introductory video
- [x] Table of contents
- [x] About tscircuit
- [x] Example circuits
- [x] Getting started 
- [x] More features
- [x] FAQ
- [x] Development Sub-Projects / Organization

Added CONTRIBUTING md file! ✅

## Issue Ticket : 

fixes #482 

## Screenshots Before and After: 

# [... cut]
closed 2024-12-11 by seveibar