tscircuit/tscircuit #483: [Docs]: Add Contributing md file

samyakshah3008, 2024-12-08:
## Description: 

- Currently, we don't have a contributing file, but it's great to have a dedicated CONTRIBUTING md file so that while we welcome new contributors for TSCircuit 💙⚡, we can directly send them this md file for better experience and we don't have to answer repeated questions in community server. 

Here is my initial implementation plan, would love to know the feedbacks! 

# Contributing to TSCircuit:

If you plan to contribute to TSCircuit, please take a moment to feel awesome ✨ People like you are what open source is about 💙. Any contributions, no matter how big or small, are highly appreciated.

## Before getting started

- Before jumping into a PR be sure to search [existing PRs](https://github.com/tscircuit/tscircuit/pulls) or [issues](https://github.com/tscircuit/tscircuit/issues) for an open or closed item that relates to your submission.
- Select an issue from [here](https://github.com/tscircuit/tscircuit/issues) or create a new one
- Consider the results from the discussion on the issue

// Todo ask about shall we introduce contributor license agreement? (Please let me know if we should add this or not)

- Accept the [Contributor License Agreement](#) to ensure we can accept your contributions.

## Taking issues

Before taking an issue, ensure that:

- The issue is clearly defined and understood
- No one has been assigned to the issue
- No one has expressed intention to work on it

You can then:

1. Comment on the issue with your intention to work on it
2. Begin work on the issue

Always feel free to ask questions or seek clarification on the issue.

## Developing

The development branch is <code>main</code>. All pull requests should be made against this branch. If you need help getting started, [join us on Discord](https://tscircuit.com/community/join-redirect).

1. [Fork](https://help.github.com/articles/fork-a-repo/) this repository to your
   own GitHub account and then
   [clone](https://help.github.com/articles/cloning-a-repository/) it to your local device.
2. Create a new branch:

- Create a new branch (include the issue id and something readable):

  ```sh
  git checkout -b feat/doc-999-somefeature-that-rocks
  ```

You are all set then to send a PR! If you get any issues or have some questions, feel free to ask in our [Discord Community Server](https://tscircuit.com/community/join-redirect).

seveibar, 2024-12-08:
We don't accept this type of contribution except from frequent contributors because otherwise our 161+ repos get spammed with people contributing getting started guides

seveibar, 2024-12-08:
Might have been a bit too harsh on this- I'm not strictly opposed to a CONTRIBUTING.md file for this repo, but I think it needs to be very specific on how to get started, e.g. it should link to https://blog.tscircuit.com/p/how-tscircuit-works-compiling-functional and our bounty dashboard https://algora.io/org/tscircuit/bounties/community

samyakshah3008, 2024-12-08:
It's completely fine and I get it 😄 but this is really kind approach from you, this shows how TSCircuit cares about new comers 💙💙
Got it @seveibar , I'll prepare a specific content on how to get started and will take feedbacks for the same! ⚡

4b11b4, 2025-03-28:
After reviewing 
https://algora.io/org/tscircuit/bounties/community

I'm wondering if some bounties could be cleaned up? 

For example, I'd take on the nano/uno bounties, but my gut says these are old and should be closed?