tscircuit/tscircuit #487: net labels in schematic are not having same names

imrishabh18, 2024-12-17:
In the case of `M1_pin13` which is connected to the capacitors `C1` and `C3`, is easy to understand by the net label. But in case for the trace connecting the `JP1_pin8` to the `M1_pin24`, it would be better if we use either of the net label name for both the ports and make it same. Similar to the `M1_pin13` case.

snippet - https://tscircuit.com/editor?snippet_id=d2c1c529-97b0-41de-96fa-2cc0c84ba342
[image]

cc: @seveibar

seveibar, 2024-12-17:
Yes they must have the same name, cc @Abse2001 

Prefer the chip with the lowest renderId

Abse2001, 2024-12-21:
/attempt #487

heysujal, 2024-12-30:
@imrishabh18 I understand `M1_pin13` is being used 2 times and labels are shown seperately for them. But, how else would we know that they were connected?
I am new to this project, does this issue require any pre-requisite like knowledge of circuits?

AbhinavTheDev, 2025-01-26:
/attempt #487

Abse2001, 2025-01-26:
@seveibar I think you fixed this issue, if so we should close it?

closed 2025-03-02 by Anshgrover23