tscircuit/tscircuit #508: Support Previewing Circuit Json in Browsers without using React

seveibar, 2025-01-24:
This feature would not take very long to build, but I want to make sure it has some interest before building it. Please upvote if you're interested!

If you have a specific framework you'd like support for, also please let me know in a comment! By default we will release a vanilla JS library!

Anshgrover23, 2025-01-24:
@seveibar So do we want to create circuit in vanilla js as well like we do in react, is this right?

DrSensor, 2025-02-02:
me: rescript without react

seveibar, 2025-02-03:
@Anshgrover23 yes, it's actually not to hard to do if we e.g. bundle runframe w/ React. We'll probably always internally run react, but we can make it so that the user doesn't have to deal with react at all via special bundling!

seveibar, 2025-02-15:
this is now possible using https://runframe.tscircuit.com/iframe.html (see runframe docs https://github.com/tscircuit/runframe )