tscircuit/tscircuit #509: Add Manual Arrow Navigation to Trending Snippets

Ayushjhawar8, 2025-01-25:
## Description

Right now, the "Trending Snippets" scrolls automatically, which is great, but there’s no way for users to control it manually. This can be a bit frustrating for those who want to browse at their own pace instead of waiting for the animation to scroll on its own.

## Reference

[image]

seveibar, 2025-01-25:
We should just create a dedicated page for trending snippets

bikaxh01, 2025-03-15:
hey i would like to contribute in this
Here is what i understand....
1. add arrows on left and right side for scrolling 
2. we need to create dedicated trending page where all the trending snippets are shown in card