tscircuit/tscircuit #600: issue with routing? unsolvedConnectionsMap.entries().map is not a function

cemag1, 2025-04-15:
Hey all,

facing an error starting with "unsolvedConnectionsMap.entries().map is not a function" 

[image]

As a result the traces are not shown in the pcb (though they do show in the schematic).

Any clue how to solve it?

Sorry if this has been asked before - I did my best to find similar Qs with no luck

seveibar, 2025-04-16:
Hey @cemag1 , thanks for the issue!!

I haven't seen this before but it looks related to the autorouter, are you able to post any of the code that created this error?

cemag1, 2025-04-16:
Hey @seveibar thank you for the ongoing efforts to build such a useful tool! 

From my experience it occurs for any circuit, even the ones in the trending snippets section;

[image]
[image]

As you can see the routes are missing in the pcb view.

imrishabh18, 2025-04-16:
@seveibar This seems like a safari issue, I tried it in safari and I can reproduce this issue

[image]

@cemag1 I would recommend you to use Chrome or some other browser for right now, until we figure out this thing

cemag1, 2025-04-16:
Interesting observation, thanks! I will leave open for now.

imrishabh18, 2025-04-16:
@cemag1 Another thing which you can possibly do is use our cloud autorouter which works fine in Safari. Just add the `autorouter="auto-cloud"` at the root of the `<board />`

[image]

seveibar, 2025-05-02:
Just an update, Safari now has this feature (in any version 18.4+)

[image]

I'm hesitant to add shims to the autorouter because I'm worried about performance issues. I'd prefer to wait this out but let's see if the safari rollout takes longer than expected