tscircuit/tscircuit #616: schematic not showing up

marsimgit, 2025-05-01:
I come to the (docs.tscircuit.com/tutorials/building-led-matrix) web page from hackaday article link.

In the web page the Schematic Circuit Preview shows the missing image icon, no circuit is displayed.

seveibar, 2025-05-01:
@marsimgit which browser are you using?

[image]

marsimgit, 2025-05-01:
👍 

I was using MS Edge

seveibar, 2025-05-01:
this is now fixed, we auto-deploy when there are updates to tscircuit sub-packages and our image server auto-updated without running test workflows, we've patched it so we don't deploy without running tests and have fixed the underlying issue (which had to do with our transition to a dedicated parts engine/BOM API module https://github.com/tscircuit/eval/pull/259)

1. Tests are now running on CI before version updates are deployed to svg.tscircuit.com https://github.com/tscircuit/svg.tscircuit.com/pull/167
2. We are going to add status checks to monitor these doc images in the future https://github.com/tscircuit/status/issues/21

@marsimgit closing this for now, again, THANK YOU! It could have been a day or so until I noticed this.

[image]

closed 2025-05-01 by seveibar