tscircuit/tscircuit #620: Multiple wires going to "same" net + proposed algo

vekexasia, 2025-05-06:
As per Discord discussion the current autorouter seems to wire multiple "elements" to the same location.  you can see it here below. A probably valid DRC board with overlapping paths going same direction.

[image]

This will overcomplicate wiring `wasting copper` :) For example we might end up in a situation like the following (bottom layer not considered for clarity)

[image]

What you usually would like to obtain is something like this:

[image]

It will allow for much simpler solutions to high density boards....

An heuristic in pseudocode to create something similar to the desired state would be:

```typescript
let unsolvedNodes = nodes.sortBy("numOfEdges");
while (unsolvedNodes.length >0) { 
  let curNode = unsolvedNodes.pop();
  let nodesToConnect = curNode.neighbors();
  let all = [curNode, ...nodesToConnect]
  let L = all.length;
  // remove all neighbors and assume solvability
  for ( const n of nodesToConnect) { 
    unsolvedNodes.remove(n)
  }
  // path find matrix calculation
  let paths = Matrix(L * L)
  for (let i=0; i<L; i++) {
    for (let j=i+1; j<L; j++) {
      paths[i][j] = pathFind(all[i], all[j]);
    }
  }
  // calculate MST ( minimum spanning tree )
  res = kruskalMST(paths)
  if ( res is false ) { throw Error }
}
```

This algorithm as it is should be `O(N^2 * O(pathFind))` where N are the connection points/smd pads or `nodes` in the algo before.

IT should already produce a good pcb ready to be optimized later in the pipeline.

Further optimizations:

- Selecting the `pathFind` step is crucial but I wouldnt be surprised if a simple A* produces a good starting point. furthermore we could potentially cache `weights` from previous steps as `all[i]` is being used as target for all the subsequent nodes.  
- instead of calculating full matrix we could heuristically filter  out too far  (air/linear distance) nodes.

vekexasia, 2025-05-07:
I had some time to think about the algorithm above and i think there are some improvements to be made. 

But first this algorithm by using a MST solver won't allow for wires connecting to other wires to share a path. aka:

[image]

this situation will be solved like this but won't allow something like:

[image]

Since this is a decent limitation of the algorithm proposed above I think that even if it should improve the current reported situation, some extra steps need to be made in order to allow path-sharing.

I am cooking something but i am not ready yet to propose anything

vekexasia, 2025-05-07:
New algorithm below. 

```typescript

// calculate node evaluation priority
calculatePriorities(nodes);
let unsolvedNodes = nodes.orderBy("priority", "asc");
while (unsolvedNodes.length > 0) {
  const curNode = unsolvedNodes.pop();
  const nodesToConnect = curNode.neighbors;
  const all = [curNode, ...nodesToConnect];
  const L = all.length;

  // path find matrix calculation
  const paths = Matrix(L * L);
  for (let i = 0; i < L; i++) {
    for (let j = i + 1; j < L; j++) {
      paths[i][j] = pathFind(all[i], all[j]);
    }
  }
  // calculate MST ( minimum spanning tree )
  let mst = kruskalMST(paths);

  optimizeMST(msg)

  // remove all neighbors and assume solvability
  for (const n of nodesToConnect) {
    unsolvedNodes.remove(n);
  }

  // update priorities for remaining unsolved nodes
  calculatePriorities(unsolvedNodes);
  unsolvedNodes = unsolvedNodes.orderBy("priority", "asc");
}
```

## priorities?

Yes nodes should be evaluated in a specific order. The algorithm does not strictly enforce the order. As a matter of fact there might be different ordering strategies based on the board charateristics.

For exammple:

 - does the node have many connections?
 - does it handle timingcritical stuff? is it a clock signal?
 - is itanalog or differential pair?
 - is it BGA port?
 - is node in a dense area?
 - is it power trace?
 - does it require impedanceControl? (aka antennas stuff?)
 - ...

I'd assign a score to each of these (and others). it should help us route critical nodes first. 

It could also be worthwhile exploring the idea of assigning priority buckets instead of discrete values.

## priority reodering at each loop?

yes. when a tree is being routed the pcb changes hence possibly & probably altering some (if not all) the remaining nodes scores. Ex: after routing a clock trace, some of the elements acquire some score for the `inDenseArea` attribute.

## What about MST and path optimizing?

The `optimizeMST` should take an MST. From Wikipedia:

````
A minimum spanning tree (MST) or minimum weight spanning tree is a subset of the edges of a connected, edge-weighted undirected graph that connects all the vertices together, without any cycles and with the minimum possible total edge weight.
```

So the MST optimizes the weight for the overall tree but we also want to reduce trace length and number of vias. so the optimizer should try to see if by connecting an existing path to another different path, it gets a better score.

for example consider the following optimized version (top) against its MST

![Image](https://github.com/user-attachments/assets/7e1950ec-0c2d-46ea-8f6d-1fc09b5029c5)

what the optimizer should aim for is (for example) replacing some of the paths (red arrows)

seveibar, 2025-05-08:
Oh interesting, just seeing this now, have you heard of a steiner tree? The algo you're proposing looks similar