tscircuit/tscircuit #692: `group.TODO_REMOVE_THIS`

shimaore, 2025-06-26:
See https://tscircuit.com/stephane-alnet/simple1#files for a full example

This fails on the CLI and when taking the project to the tscircuit.com Editor
```
Build failed: Error: Could not find port for selector "group.TODO_REMOVE_THIS > chip.U1 > port.VDD". Component "group.TODO_REMOVE_THIS > chip.U1 > port" not found
```

shimaore, 2025-06-26:
Sidenote for anyone encountering this: using `<group name="some-name-here">` instead of an anonymous group works properly.

seveibar, 2025-07-06:
CC @imrishabh18 is this fixed? Thanks for the report @shimaore

imrishabh18, 2025-07-06:
Yes, seems to be fixed. Just tried the reproduction which @shimaore shared on `tscircuit.com`

closed 2025-07-06 by seveibar