tscircuit/tscircuit #758: Add cadModel.gltfUrl support

seveibar, 2025-09-05:
- add in tscircuit/props
- add in tscircuit/circuit-json
- implement in tscircuit/core
- implement model loader in tscircuit/3d-viewer

You must be assigned this task.

/bounty $150

ShiboSoftwareDev, 2025-09-06:
@seveibar cli too right?

and can you please explain a little about circuit-json-to-gltf?

jayantpranjal0, 2025-09-28:
/attempt #758

seveibar, 2025-09-29:
The last checklist item is so difficult to implement, I've created a separate bounty for it here: https://github.com/tscircuit/circuit-json-to-gltf/issues/14

We're going to close this issue and let that issue take over. @ShiboSoftwareDev i think you should have the claim (idk how to award you though)

ShiboSoftwareDev, 2025-09-29:
@seveibar you can close this issue and give me a tip on one of my related prs, I've been doing work towards the final issue but to no success. I managed to get the model loaded and it worked but it broke the model-viewer used in the cosmos preview

ShiboSoftwareDev, 2025-09-29:
The problem is it breaks silently with no errors or warnings lol