tscircuit/tscircuit #770: Support "knockout" silkscreen text

seveibar, 2025-09-10:
Knockout silkscreen text is basically a filled rect with the text omitted. It is a cool visual appreance.

- [ ] Propose parameters for creating knockout silkscreen text to tscircuit/circuit-json
- [ ] Implement in circuit-to-svg
- [ ] Implement in pcb-viewer
- [ ] Implement in 3d-viewer (NOTE: should probably use circuit-to-svg internally, may be larger issue, i'm not sure we show any silkscreen)
- [ ] Allow specifying knockout in tscircuit/props

## Example Knockout Silkscreen

[image]

[image]

/bounty $150

IsxImattI, 2025-09-16:
/claim

h30s, 2025-09-24:
/attempt #770

bimakw, 2026-01-21:
/attempt #770

Hey, I can finish this one. I did some analysis and here's the current state:

**Already Done:**
- `circuit-json`: `is_knockout` + `knockout_padding` fields exist in `pcb_silkscreen_text.ts`
- `props`: `isKnockout` + padding props exist in `silkscreenTextProps`
- `circuit-to-svg`: Knockout already works for `pcb_copper_text` with SVG mask technique

**Remaining Work:**
1. **circuit-to-svg**: Port knockout rendering from `create-svg-objects-from-pcb-copper-text.ts` to `create-svg-objects-from-pcb-silkscreen-text.ts`
2. **core**: Verify SilkscreenText component maps knockout props to circuit-json (add if missing)
3. Add tests for knockout silkscreen text

The copper-text implementation uses SVG masks which is the right approach - I'll adapt it for silkscreen with the appropriate color handling.

Will have PR ready soon.

bimakw, 2026-01-22:
/attempt #770

I'll implement knockout silkscreen text support. Plan:

1. **circuit-json**: Add `knockout` boolean prop to `pcb_silkscreen_text` schema
2. **circuit-to-svg**: Render knockout as filled rect with text mask/clip-path
3. **pcb-viewer**: Add knockout rendering support
4. **tscircuit/props**: Add `knockout` prop to silkscreen text component

Starting with circuit-json schema changes, then moving to circuit-to-svg rendering. Will submit PRs incrementally.

bimakw, 2026-02-09:
/claim #770

Knockout silkscreen text is implemented and merged:

- **circuit-to-svg#506** — SVG knockout rendering with mask (merged Feb 6)
- **core#1886** — map isKnockout + knockoutPadding props to circuit-json (merged Feb 7)
- **circuit-to-canvas#159** — knockout rendering for pcb-viewer (submitted)

All checklist items covered except 3d-viewer which likely uses circuit-to-svg internally.

St34lthcole, 2026-02-16:
/attempt

Comprehensive test coverage PR for knockout silkscreen text: https://github.com/tscircuit/circuit-to-svg/pull/510

The implementation already exists in circuit-to-svg. This PR adds 28+ test cases (1043 lines) addressing the reviewer feedback from PR #815 that requested thorough tests.

Abse2001, 2026-02-27:
@li224242 we already fixed this issue