tscircuit/tscircuit #785: Add support for `symbol={<symbol>...</symbol>}` and demonstrate with a basic symbol featuring lines, rects, circles, arcs

seveibar, 2025-09-13:
- [ ] Introduce Symbol.ts (similar to CadAssembly.ts or Footprint.ts)
- [ ] Modify NormalComponent.ts to handle symbol rendering
- [ ] Handle `symbol="somestring"` which is an alias for `symbolName="..."`
- [ ] Introduce primitives as needed in tscircuit/props

/bounty $200

MAVRICK-1, 2025-09-13:
@seveibar we want make the changes in other repo ?

BenraouaneSoufiane, 2025-09-16:
@seveibar I see @imrishabh18 removed the label, so it's not bounty anymore?

imrishabh18, 2025-09-16:
@BenraouaneSoufiane The bounty has been removed before you had raised PR or added a comment of `attempt`. But since I have not yet raised the PR and you already have one. I think we can tip you for the work you have done if your PR get's merged

BenraouaneSoufiane, 2025-09-16:
@imrishabh18 Thank you for your reply, I've finished it.

imrishabh18, 2025-09-17:
@BenraouaneSoufiane Your PR's are confusing it doesn't even make sense. It's just full of AI slop, please review your own PR's before requesting a review.