tscircuit/tscircuit #788: Support "pinout diagram" output format

seveibar, 2025-09-15:
- [ ] Add support for `pcb_port.is_board_pinout` in circuit-json (easy)
- [ ] Add support for `pinAttribute` type `includeInBoardPinout` in tscircuit/props (easy)
- [ ] Make adjustments in tscircuit/core to pass along relevant attributes to internal circuit json (easy)
- [ ] Create new output type in `circuit-to-svg` that renders more realistic board SVG as "pinout" type (hard). Similar to assembly SVG but with more colors.

Example of appearance:

[image]

/bounty $400

sparkybug, 2025-09-17:
/attempt #788

gratedayne, 2025-09-30:
/attempt #788

I will implement the "pinout" output format by:
1. Adding support for pcb_port.is_board_pinout in circuit-json
2. Adding pinAttribute.type.includeInBoardPinout in tscircuit/props
3. Modifying tscircuit/core to pass these attributes to the circuit JSON
4. Creating a new SVG renderer in circuit-to-svg that:
   - Draws components as rectangles
   - Labels each pin with text
   - Colors pins by type (power, ground, signal)
   - Uses a clean, board-like layout similar to assembly view but with enhanced pin labeling
I will provide a demo video showing the output.

closed 2025-10-03 by seveibar