tscircuit/tscircuit #803: Implement the "Automerge Pipeline"

seveibar, 2025-09-19:
Current, @tscircuitbot opens a pull request automatically on tscircuit/eval anytime something is merged to tscircuit/core. This is good, but we want to extend the automerge pipeline, the following dependencies triggers should be set up

- [Workflow to create PR](https://github.com/tscircuit/core/blob/96019153c07f7c6b8ddd6d8bd192b8a11d8dcdf4/.github/workflows/bun-pver-release.yml#L34)
- [Workflow to automerge (not working totally properly)](https://github.com/tscircuit/eval/blob/main/.github/workflows/auto-merge-tscircuitbot.yml)

Automerging also needs to work, @ArnavK-09 has attempted this on eval but it's not completely working yet. Tests must pass to automerge. We should implement as a "ploppable" workflow via [tscircuit/plop](https://github.com/tscircuit/plop) i.e. **the workflow file should be the same for every repo with only minor customization of the name of repos, no fragile per-test configuration**

```yaml
REPO: tscircuit/eval
TRIGGER: tscircuit/core merge
ACTION: tscircuit/core update

REPO: tscircuit/runframe
TRIGGER: tscircuit/eval merge
ACTION: tscircuit/eval update

REPO: tscircuit/runframe
TRIGGER: tscircuit/3d-viewer merge
ACTION: tscircuit/3d-viewer update

REPO: tscircuit/runframe
TRIGGER: tscircuit/pcb-viewer merge
ACTION: tscircuit/pcb-viewer update

REPO: tscircuit/runframe
TRIGGER: tscircuit/schematic-viewer merge
ACTION: tscircuit/schematic-viewer update

REPO: tscircuit/cli
TRIGGER: tscircuit/runframe merge
ACTION: update tscircuit/runframe

REPO: tscircuit/tscircuit
TRIGGER: tscircuit/cli merge
ACTION: update tscircuit/core, tscircuit/eval, tscircuit/runframe, tscircuit/cli
```

DO NOT INTRODUCE CIRCULAR DEPENDENCIES! The triggers above are carefully designed to avoid circular dependencies while still doing a full update

```mermaid
flowchart TD

    %% Core to Eval
    core[tscircuit/core] -->|merge| eval[tscircuit/eval]

    %% Eval to Runframe
    eval -->|merge| runframe[tscircuit/runframe]

    %% 3D Viewer to Runframe
    viewer3d[tscircuit/3d-viewer] -->|merge| runframe

    %% PCB Viewer to Runframe
    pcbviewer[tscircuit/pcb-viewer] -->|merge| runframe

    %% Schematic Viewer to Runframe
    schemaviewer[tscircuit/schematic-viewer] -->|merge| runframe

    %% Runframe to CLI
    runframe -->|merge| cli[tscircuit/cli]

    %% CLI to Tscircuit
    cli -->|merge| tscircuit[tscircuit/tscircuit]

```

seveibar, 2025-09-19:
/bounty $250 takes a lot of PRs so you must be assigned

Trex96, 2025-09-20:
Hey! @seveibar 

I want to tackle this automerge issue.
I can contribute in :

- Built similar GitHub Actions workflows before
- Uderstand your dependency DAG prevents circular deps
- Can create the reusable plop template you need

My approach:

- Fix @ArnavK-09's automerge issues (proper test gates + merge conditions)
- Build single workflow template with minimal config (just repo names)
- Deploy across all 6 repo triggers in your chain
- Test the full pipeline: core → eval → runframe → cli → tscircuit

I've already studied tscircuitbot's current PR logic - I know exactly what's breaking the automerge flow. Ready to start immediately.
Can you assign this to me?

seveibar, 2025-09-26:
We figured this one out @Trex96 , @imrishabh18 is implementing a solution

naaa760, 2025-09-27:
/attempt #803
Hello @seveibar! would you please assign me for this problem?

seveibar, 2025-09-27:
Hey @naaa760, @imrishabh18 is already fairly far into his implementation. Ive removed the bounty to make it more clear it got grabbed by staff. Hopefully there’s another issue you can find!